openhwfoundation/cv32e40x #637: Documentation: load/store error

eroom1966, 2022-08-03:
**Component:Doc **:

The documentation seems to use the same expression to determine a load or store error
data_err_i = 1 and data_rvalid_i = 1 for load
data_err_i = 1 and data_rvalid_i = 1 for store

I presume rvalid should be 0 in the store error case

Silabs-ArjanB, 2022-08-03:
The expressions are correct. The differentiation is made by the 'for load' or 'for store' part of the sentence. Every OBI transaction is either a read (data_we_o = 0) or a write (data_we_o = 1). These signals are only valid on the boundary of the core during the address phase of a data transaction. The core itself remembers internally if a data_rvalid_i that is eventually returned is 'for a load/read' or 'for a store/write) and it can classifiy the data errors correctly according to its internal state at the time that the error is received. This tracking logic can easily be build based on just external OBI signals as OBI response phase transfer remain in the same order as their corresponding address phase transactions.

PilotAware, 2022-08-03:
@silabs-hfegran 
Henrik, what is the proposed logic for telling the RM that either of the NMI bus errors have occurred to set either 1024/1025 as the cause

Silabs-ArjanB, 2023-03-13:
@silabs-mateilga Can this issue be closed now?