openhwfoundation/cv32e40x #644: Documentation: rvfi_trap size declaration

eroom1966, 2022-08-19:
in the documentation the rvfi_trap is documented as having width 14 bits  ([13:0])
however the declaration is 12 bits
    input logic [TRAP_WL-1:0]        rvfi_trap,
TRAP_WL is defined
localparam TRAP_WL          = 12;

so the declaration is
input logic [11:0]        rvfi_trap,

Silabs-ArjanB, 2022-08-19:
Hi @eroom1966 , the documentation is right in this case. In what file do you see the 'input logic [TRAP_WL-1:0] rvfi_trap,' declaration. I don't think that is actually inside this (cv32e40x) repos as here the signal is declared as output (not input) with type 
rvfi_trap_t, which is 14 bits and is defined in https://github.com/openhwgroup/cv32e40x/blob/master/bhv/include/cv32e40x_rvfi_pkg.sv.

Maybe this ticket needs to be moved to the core-v-verif repos?

eroom1966, 2022-08-19:
Hi @Silabs-ArjanB 
this is in
core-v-verif/lib/uvm_agents/uvma_rvfi/uvma_rvfi_constants.sv

I am looking at the cv32e40x/dev branch

Line 25
https://github.com/openhwgroup/core-v-verif/blob/cv32e40x/dev/lib/uvm_agents/uvma_rvfi/uvma_rvfi_constants.sv

thx
Lee

eroom1966, 2022-08-19:
> Maybe this ticket needs to be moved to the core-v-verif repos?

I think you are right, is there an easy way to do that ?