openhwfoundation/cv32e40x #653: [XIF] LSU may send mpu_status to WB too early

silabs-oysteink, 2022-08-25:
If there is an offloaded instruction in WB waiting for xif_result_valid, and the MPU blocks a load or store in EX, the lsu_mpu_status_wb may get updated with the LSU instruction related fault before the offloaded instruction is completed.

The MPU does not get backpressure from wb_ready, but rather the LSU internal counter that counts outstanding transactions. When X_EXT=0 this works, but with X_EXT=1 the MPU must take wb_ready into account.

Silabs-ArjanB, 2023-01-26:
Removing resolved label as applied fix does not actually address the stated issue. The MPU status can still get updated too early. In the applied fix it is just kept valid longer. Need to reconsider fix strategy.

Silabs-ArjanB, 2023-01-26:
The LSU as a whole (including watchpoint unit and MPU) should not need to deal with back pressure from WB simply because the OBI response data itself can also not deal with back pressure (as rready=1 always). Therefore the WB stage needs to be receptive for 1-cycle (non-sticky) outputs from the LSU always only assuming that the OBI protocol is being followed (e.g. earliest rvalid the cycle after granted request and responses are in order). (Therefore potentially additional buffering or priority logic needs to be added in WB).