openhwfoundation/cv32e40x #709: NMI is taken during debug single step

silabs-mateilga, 2022-11-30:
**Component:RTL**
NOTE: Bug was found on CV32E40S during development of the debug assertion set, so reproduction is only possible with 40s assets. The feature is shared and will be fixed on the 40X, which is the reason the issue is filed here.

### Description
An NMI firing during debug, indicated by dcsr.NMIP, is taken when dret instruction retires with dcsr.STEP high.

### Steps to Reproduce 
Run formal verification on CV32E40S assert a_step_irq_dis in uvmt_cv32e40s_debug_assert.sv
NOTE: This assert is not PRed at the time of writing, but is available here: https://github.com/silabs-mateilga/core-v-verif/tree/debug_assert_updates

Silabs-ArjanB, 2022-11-30:
@silabs-mateilga What is the value of dcsr.stepie when this happens?

silabs-mateilga, 2022-11-30:
Stepie was mistakenly not taken in to account in the assert, the bug is not real.

closed 2022-11-30 by silabs-mateilga