openhwfoundation/cv32e40x #711: NMI handler entry retires during single stepping

silabs-mateilga, 2022-12-01:
<br>**Note**: This issue is found during work with expanding the debug assertion set in the 40s/dev branch of core-v-verif, and is discovered on the cv32e40s rtl. The issue is still created here, as the debug features are developed on the 40X and merged to 40S.  

**Component:RTL**: For issues in the RTL (e.g. for files in the rtl directory)

###Description
Upon a dret from debug mode with dcsr.STEP and dcsr.STEPIE set, trap handler entry should not retire before reentering debug mode. Formal verification of assert a_step_no_trap gives a CEX that proves this is violated in the event of a Load Bus Fault NMI, other interrupt causes can probably give the same problem.

### Steps to Reproduce
Formally verify the debug_assert/a_step_no_trap assertion found in this PR of core-v-verif: [#1515](https://github.com/openhwgroup/core-v-verif/pull/1515)

Silabs-ArjanB, 2022-12-01:
Hi @silabs-mateilga Are you sure that the assertion is correct and not giving a false positive on the following scenario?

[image]

silabs-mateilga, 2022-12-01:
@Silabs-ArjanB I'm fairly sure, as @silabs-oysteink has confirmed there is an issue in rtl that should be highlighted by this assert. The scenario highlights a break of the following section in the debug spec (4.4.1) as far as I understand: 
[image]

silabs-mateilga, 2022-12-12:
Assert passes

closed 2022-12-12 by silabs-mateilga