openhwfoundation/cv32e40x #745: Debug cause is wrong when a single step is interrupted by an NMI, and this coincide with a CEBREAK

silabs-mateilga, 2022-12-22:
When a single step operation is interrupted by an NMI, debug mode shall be reentered immediately, and the cause field should indicate "step"(0x4). However, if this coincides with another debug entry cause, in the instance shown a CEBREAK instruction, the cause field is set to "ebreak"(0x1). Other instances might also be problematic, but this has not been shown.

[image]

**Component:RTL**

### Steps to Reproduce
Only shown with under-development assertions, contact me directly if more details are required. @silabs-oysteink is aware of the details.

Silabs-ArjanB, 2023-02-06:
Should be fixed by https://github.com/openhwgroup/cv32e40x/pull/775

silabs-mateilga, 2023-02-08:
Fix confirmed in formal verification