openhwfoundation/cv32e40x #793: rvfi_nmip timing

silabs-mateilga, 2023-02-22:
At the moment rvfi_nmip is immediate set directly from controller_fsm, and does not match rvfi_valid timing, creating a mismatch between the value of rvfi_csr_dcsr.rvfi_csr_rdata and rvfi_nmip on an rvfi_valid. Is this correct, or should the rvfi_nmip be moved to the same timing as the other signals qualified at rvfi_valid.

Silabs-ArjanB, 2023-02-22:
Hi @silabs-mateilga I don't immediately recall why we did give it different timing and why the signal is there at all. How/why is this signal being used in core-v-verif?

silabs-mateilga, 2023-02-23:
I am not aware of it being used for anything specifically today, however I am trying to verify the correctness of dcsr.nmip, and this signal seemed like a natural choice for a reference.

Silabs-ArjanB, 2023-02-23:
If it is not used, then let's remove it. If you want to verify the CSRs then use the RVFI CSR signals and not this dedicated bit.

silabs-mateilga, 2023-02-23:
Signal is used in imperas_dv_wrap, and the timing difference is intentional. Closing issue.

closed 2023-02-23 by silabs-mateilga