openhwfoundation/cv32e40x #800: Redundant mem_result when using CORE-V-XIF

davidmallasen, 2023-03-08:
## Redundant mem_result when using CORE-V-XIF

### Component
Issues in the RTL

### Steps to Reproduce
Git hash: v0.8.0 974ca469f3a7598e99b5c6eb9d6a4b355fe8c9da

Hello,

I'm using CV32E40X inside [X-HEEP](https://github.com/esl-epfl/x-heep) together with a coprocessor that interacts with XIF. Through the XIF I'm receiving a redundant mem result from the CV32E40X that I believe should not be there. I'm not sure if it's a problem from my side or from the CV32E40X, but checking the waveforms I see everything correct from the coprocessor side (as I understand it).

I'm running the following snippet of code, where `plw` can be read as a normal lw but to an external register file in the coprocessor, and `peq` are equal operations between registers of this external RF that should write back a 0 or 1 to the GPRF of CV32E40X:
[image]

I atttach a screenshot of the waveform of the XIF signals:
[image]

The first 3 `issue_ready` - `issue_valid` handshake signals are for the 3 `plw`, and the next 3 are for the 3 `peq`. They are all signaled a `commit_kill = 0`, so the coprocessor initiates memory retrievals of the `plw` instructions with the 3 `mem_valid` - `mem_ready` handshakes. CV32E40X replies correctly through the `mem_result` interface, but after a while (right after the 3 `peq` instructions are executed and signaled via the `result` interface. There is an extra `mem_result_valid` where the yellow cursor is (identical to the last one) that should not be there, as there is no 4th mem request.

The output of the above program is: 
~~~
 PEQ test FAIL - Values: 7f939d17 7f6ec2bf 806e6c7b 1 1 1
~~~
Where the last 3 ones should be 0 1 1.

I also attach a VCD dump of this execution.

[waveform.vcd.zip](https://github.com/openhwgroup/cv32e40x/files/10922588/waveform.vcd.zip)

If it helps, this only happens with operations that end up writing back to the GPRF. Other snippets of code such as the following, work correctly.
[image]

Thanks!
David

Silabs-ArjanB, 2023-03-09:
Hello @davidmallasen Thank you very much for filing this issue. It looks like an RTL bug indeed. Unfortuantely the vcd file does not seem to actually contain data for the cv32e40x_core_i scope and below:

[image]

Is it possible for you to generate a new vcd tracing all signals below that scope?

Best regards,
Arjan

davidmallasen, 2023-03-09:
Hello @Silabs-ArjanB 
I'm sorry, but I'm able to see all those signals in GTKWave. Maybe the issue is that I only included the important time interval?

[image]

davidmallasen, 2023-03-09:
That may be a limitation on the SV interfaces, I can't see it either

davidmallasen, 2023-03-09:
Great, thanks @Silabs-ArjanB !

Silabs-ArjanB, 2023-03-09:
[image]

davideschiavone, 2023-03-09:
hi @Silabs-ArjanB - it seems you commented without text :D

Silabs-ArjanB, 2023-03-09:
Yes, just a screenshot of the relevant signals as a reminder to myself

Silabs-ArjanB, 2023-03-09:
@davidmallasen Are you blocked by this?

davidmallasen, 2023-03-09:
Yes, it's not allowing me to continue with my work

davidmallasen, 2023-09-07:
Correct, the closing of the PR was an unintended side-effect, sorry about that @Silabs-ArjanB