openhwfoundation/cv32e40x #814: [XIF] Issue_req.id not updating and missing commit_valid

davidmallasen, 2023-03-24:
## Issue_req.id not updating and missing commit_valid

### Component
Issues in the RTL

### Steps to Reproduce
Git hash: b66765acf650493901b6d3e7b44eb37bd2598f4a

Hello,

In the same setup as issue #800 , the `id` of the instruction through the `issue_req` interface is not updating (is always 0) and therefore the second time an instruction is offloaded, it gets stuck waiting for a `commit_valid` that never arrives.

I'm running this snippet of code, where it gets stuck after the issue of the second load instruction:
[image]

Firstly there is a random offloading that gets killed instantly:
[image]

Then the first load instruction offloads and has to wait for the `mem_ready`:
[image]

Then the `mem_ready` arrives, the instruction completes, and the second load is issued. However, the `commit_valid` for this second instruction never arrives because (I presume) the id is the same as the previous ones:
[image]

I attach a VDC dump of the execution.
[waveform.vcd.zip](https://github.com/openhwgroup/cv32e40x/files/11063526/waveform.vcd.zip)

Thanks!
David

Silabs-ArjanB, 2023-09-07:
Note that https://github.com/openhwgroup/cv32e40x/pull/825 was closed but its proposed fix has not been merged yet

davidmallasen, 2023-09-07:
Correct, the closing of the PR was an unintended side-effect, sorry about that @Silabs-ArjanB