openhwfoundation/cv32e40x #831: rvfi_intr.exception is kept high when an exception is followed by a trigger debug entry

silabs-mateilga, 2023-04-18:
[image]

When an exception handler entry is immediately followed by a debug entry with cause trigger, in other words, when the first instruction in an exception handler is a debug trigger hit, rvfi_intr.exception remains high at the retire of the first instruction in the debug handler. The current expectation from verification asserts is that the rvfi_intr.exception signal should follow one retire after rvfi_trap.exception, unless an interrupt is signaled. The question here is if the verification expectation is wrong, or is the RTL implementation wrong?

Reproduce by formally verifying assert `...core_i.if_stage_i.mpu_i.pma_assert_instr_i.a_eventually_mpu2obi` with latest cv32e40s/dev branch of core-v-verif, at current date.

Silabs-ArjanB, 2023-04-18:
@silabs-mateilga Isn't this the scenario explained here?

[image]

silabs-oysteink, 2023-04-18:
I think this issue is a duplicate of issue #355 on cv32e40s.
1: Core takes exception (RVFI starts tracking in_trap[])
2: First exception handler instruction gives a trigger match (IF/ID/EX is killed, wb_valid==1 to let RVFI know about the sync debug entry).

When rvfi_valid==1 for (2), rvfi_trap will indicate a debug entry while rvfi_intr will show the cause of exception from (1).  The issue arises if this first exception instruction is a POP for instance, which may have subops in multiple stages when the trigger match debug entry happens. This can cause the rvfi_intr information within RVFI to backpropagate to the IF stage, thus causing also the next rvfi_valid to carry the same rvfi_intr information.

Silabs-ArjanB, 2023-04-18:
@silabs-oysteink is there still an issue here then because 355 is marked as resolved? Is this on an older hash?

silabs-oysteink, 2023-04-18:
Issue 355 on cv3e40s was previously marked as resolved, but the label was removed as a new case came up that showed the same behaviour.

Silabs-ArjanB, 2023-04-18:
So, this is an RVFI issue (and not an RTL issue), right?

silabs-oysteink, 2023-04-18:
Yes, this is only for RVFI.

Silabs-ArjanB, 2023-04-19:
Should be fixed by https://github.com/openhwgroup/cv32e40x/pull/835 (still needs to be merged to 40S)

closed 2023-04-20 by silabs-mateilga