openhwfoundation/cv32e40x #879: [XIF] Wrong mem_result when mem_req.size and mem_req.be are modified

davidmallasen, 2023-06-28:
Hello,

I'm using CV32E40X inside [X-HEEP](https://github.com/esl-epfl/x-heep) together with a coprocessor that interacts with XIF. I want to load 16-bit data instead of 32 bit. For that, I set the `mem_req.size` to `1` and `mem_req.be` to `0011`. However, in the `mem_result.rdata` I receive 32-bit data.

In the following example, I'm trying to load first a memory address which contains `0x7F94` and then another one which contains `0x7F6F` and a third with `0x806E` (consecutive in memory). However, I get as a result `0x7F947F6F` in the first two loads and `0x806E0000` in the third load.

[image]

I also attach the vcd file of the execution which starts at 5881570 ns.

[dump.zip](https://github.com/openhwgroup/cv32e40x/files/11896197/dump.zip)

From this, I suspect the core is not responding to the `mem_req.size` and `mem_req.be` signals. Any input on this?

David

Silabs-ArjanB, 2023-06-29:
Hi @davidmallasen , without having looked at the attached vcd file I wonder if the following part from the CORE-V-XIF specification explains the behavior you see:

[image]

(On purpose we leave the data received on the CPU OBI interface untouched and forward it to the coprocessor. The CPU would have no idea whether to for example 0-extend or sign-extend partial width transfers, etc.)

davidmallasen, 2023-06-29:
Hello @Silabs-ArjanB . Exactly. Since I'm setting the `be` signal to `4'h3` I would expect valid data only on the 2 least significant bytes out of the 4. However, the coprocessor receives the full 32-bit word transfer with data in whichever part of the word (either most significant 2 bytes or least significant 2 bytes depending on whether it is 2-byte aligned or 4-byte aligned in memory). Note that the `mem_req.addr` in the request also increments in 2 bytes, not 4.

Do you mean that the OBI interface is the one in charge of complying with the `mem_req.size` and the `mem_req.be` signals? Maybe there is the problem? That's where I got running down the signal rabbit-hole in the CPU, but I'm not sure.

Silabs-ArjanB, 2023-06-29:
Hi @davidmallasen If you just look at the OBI interface and at the memory result interface (and not at any signals inside the CPU), can you confirm that the 32 bit rdata on the OBI interface is identical to the 32 bit rdata on the memory result interface? 

Are you doing something like setting be to 4'h3 while setting the address to something that has addr[1:0] = 0x2? (That would be illegal)

davidmallasen, 2023-06-29:
You are correct. I missed the dependency between `be` and `addr` that is at the end of the paragraph you mentioned in your previous response. Thanks!

davidmallasen, 2023-06-29:
Hello again @Silabs-ArjanB . I have changed the signals to comply with the `be` and `addr` dependencies (by setting `be` to `4'b1100`) and the result is still the same as can be seen in the below screenshot. I can confirm that the 32-bit `rdata` on the data OBI interface is identical to the 32-bit `mem_result.rdata` of the XIF when `mem_result_valid` is asserted.
[image]

Silabs-ArjanB, 2023-06-29:
Hi @davidmallasen Given that OBI rdata and XIF result rdata are equal, doesn't that mean that the issue is in whatever circuit returns the data onto OBI rdata?

davideschiavone, 2023-06-29:
Hi @Silabs-ArjanB - I think I may have found the issue.

in `cv32e40x` I see [this](https://github.com/openhwgroup/cv32e40x/blob/master/rtl/cv32e40x_load_store_unit.sv#L310) piece of code:

```
if (xif_req) begin
        lsu_size_q     <= 2'b10;
        lsu_sext_q     <= 1'b0;
        lsu_we_q       <= xif_mem_if.mem_req.we;
        rdata_offset_q <= '0;
```

So I guess @davidmallasen expected to have the data aligned whereas he always gets the full word from the memory (which is correct) - so I guess the co-processor needs to implement the alignment of the data?
Would it be the same for misalignment memory accesses?

Do we have this written somewhere?

Silabs-ArjanB, 2023-06-29:
@davideschiavone This behavior is what I marked in yellow in my first reply above (which is taken from the CORE-V-XIF spec).

>> so I guess the co-processor needs to implement the alignment of the data?

Yes

>> Would it be the same for misalignment memory accesses?

Note that there is a difference between misaligned memory accesses and split memory accesses. Doing misaligned memory accesses by a coprocessor can be attempted but the CPU's PMA might return an error. There is a lot of explanation in the XIF spec about this. One important remark there is that "Specifically if an unaligned memory operation crosses a X_MEM_WIDTH boundary, then it shall be broken into multiple transactions on the memory (request/response) interface by the coprocessor."

davidmallasen, 2023-06-29:
In the case that the coprocessor has to implement the alignment of the data, what would be the purpose of the `be` signal in the XIF?

Silabs-ArjanB, 2023-06-29:
It needs to be used for the target to know what data to write (exactly) and what data to read. For reads (depending on the memory type additional read data might get returned). You cannot deduce that from addr and size (byte enables do not need to be contagious)

davidmallasen, 2023-06-29:
Ah ok, it's also used for the write. Thanks a lot @Silabs-ArjanB

closed 2023-06-29 by davidmallasen