openhwfoundation/cv32e40x #945: Question on how to handle CSR instructions unwillingly offloaded to the external unit.

yvantor, 2023-09-12:
In the [following code line](https://github.com/openhwgroup/cv32e40x/blob/master/rtl/cv32e40x_id_stage.sv#L711C87-L711C87), an attempt is made to offload any CSR instruction via the X-If, with the specification that the validity of the CSR instruction will be checked later in the EXE stage. I encountered this issue while executing simple instructions, such as csrr a0, mhartid, or basic csrrs instructions involving mie and mstatus to activate interrupts. The core issued an offload request externally, but no response was received, causing the execution to be blocked. To address this, I made a simple change by forwarding the csr_illegal_o signal from the cs_registers module to the ID stage and using it to forward the CSR instruction to the external coprocessor only if it is considered illegal (you can find this change in [this commit](https://github.com/yvantor/cv32e40x/commit/7e7885854154dd07bd6cff937753c4a00a1856bd)). I implemented this workaround because I found limited information on how to handle CSR requests in this scenario. I would appreciate your guidance on the proper way to handle this issue. Thank you in advance!

Silabs-ArjanB, 2023-09-13:
Hi @yvantor Thanks for your question. When performing a CSR instruction that will be handled by the CV32E40X (like for example the csrr a0, mhartid that you mentioned) the CV32E40X will indeed try to offload that instruction (because it doesn't know yet in the ID stage that it can handle the instruction itself). It is the duty off the Coprocessor to then not accept that specific instruction for offload.

So, the proposed change to the CV32E40X in the commit that you provided is not okay and this issue needs to be addressed external to the CV32E40X instead.

yvantor, 2023-09-13:
Thank you @Silabs-ArjanB for the quick reply, all clear!

closed 2023-09-13 by yvantor